sbrodeur/ros-icreate-bbb #1: Split the published output of the icreate driver into multiple messages

sbrodeur, 2016-09-08:
The irobot_create/SensorPacket message currently contains all information about the state of the robot. It would be better to use standard message types when possible. Note that odometry and motor-related messages are already implemented.

### Contact-related: custom
bool wheeldropCaster
bool wheeldropLeft
bool wheeldropRight
bool bumpLeft
bool bumpRight
bool wall
bool cliffLeft
bool cliffFrontLeft
bool cliffFrontRight
bool cliffRight
bool virtualWall

### Odometry-related:
int32 distance
int32 angle
<http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html>

### Battery-related:
uint8 chargingState
uint16 voltage
int16 current
int8 batteryTemperature
uint16 batteryCharge
uint16 batteryCapacity
bool homeBase
bool internalCharger
<http://docs.ros.org/api/sensor_msgs/html/msg/BatteryState.html>

### IR Range-related:
uint16 wallSignal
uint16 cliffLeftSignal
uint16 cliffFrontLeftSignal
uint16 cliffFrontRightSignal
uint16 cliffRightSignal
see <http://docs.ros.org/api/sensor_msgs/html/msg/Range.html>

### Motor-related: 
int16 requestedRightVelocity
int16 requestedLeftVelocity

sgcarrier, 2016-09-12:
Seperation done. Couldn't use standard msg Range for the IR sensor as surface reflectivity has much impact on values so distance in m is very hard to establish. Furthermore, investigation should be done on the battery. When in low voltage, it will not charge correctly. I suspect the robot is always in Active mode and thus cannot charge even in standby. Perhaps a review of driver.py with proper mode selection in mind should be done.

sbrodeur, 2016-09-12:
Excellent! Only one thing: I think we should use BatteryState directly instead of the new message BatteryIrobot. We don't need to publish the battery temperature, and home base or internal charging flags can be expressed by the power_supply_status variable (i.e. set to POWER_SUPPLY_STATUS_CHARGING). Can you make the changes and recommit?

sbrodeur, 2016-09-12:
For the charging issue, maybe it is related to switching between active and passive modes:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/irobot_create/bin/driver.py#L122

sgcarrier, 2016-09-12:
Done. I will investigate the mode switching further. Commented it for now. I didn't add any special cases for when its docked or not concerning the POWER_SUPPLY_STATUS. I assume the irobot serial messages recieved for charging state indicate if its being charged ( regardless if its from the dock or the cable).

sgcarrier, 2016-09-13:
Removed sensorPacket from driver.py and behaviours.py. commented out for now all behaviours related to docking.

closed 2016-09-14 by sbrodeur