sbrodeur/ros-icreate-bbb #2: Quantify the effect of the motors on the magnetometer sensor

sbrodeur, 2016-09-08:
Currently, there is a significant bias in the magnetic field recorded by the 3-axis sensor due to the proximity of the left motor. It should be assessed whether or not this affects the accuracy of the orientation estimations when the motor is static (turned off) or rotating. If yes, the IMU should be repositioned far from the motors if we cannot do any software compensation.

sgcarrier, 2016-09-12:
Motors add noise to the data but not a significant amount. However, the motors (or the calibration?) reduce the effective range of values. For example, X values vary between 0.25 and 0.75, and Y between 0.91 and 1.25 (with motors in normal positions), with no negative values ever. Z stays constant at -0.80. So the little noise added by the motors (around -+0.1) can have a big impact. 

Investigation towards enlarging the range of values is next (calibration maybe?). Also, check towards smoothing the effects of movement on values. Since, although the error caused by the motors is small, it is very rapid and spiky, so it could cause orientation issues (smoothing filter?).

sbrodeur, 2016-09-13:
IMU unit will need to be relocated on top of the robot.

closed 2016-09-13 by sbrodeur

sbrodeur, 2016-09-14:
Due to the longer wiring for the I2C bus after relocalisation, 1.8k pullup resistors were added on the SCL and SDA lines. The clock speed was reduced from 400kHz (fast mode) to 100kHz due to higher capacitance on the line (the rise time is now about 400ns).