sbrodeur/ros-icreate-bbb #6: Provide stereo calibration parameters for the RGB cameras

sbrodeur, 2016-09-13:
We don't plan to provide rectified camera images calibrated for stereo vision as part of the dataset, but we should at least provide the calibration parameters so that is can be done later (i.e. by processing the rosbags).

See: <http://wiki.ros.org/camera_calibration>

sgcarrier, 2016-09-19:
Obtained calibration information for both cameras (using ros camera_calibration). However, right camera is inverted and it would be ideal to correct that. Next step is to create a node that will invert right camera image and synchronize the 2 cameras and also update respective camera_info topics with new calibration information.

sbrodeur, 2016-09-20:
This project allows to do lossless rotations of JPEG encoded images from Python:
https://github.com/jbaiter/jpegtran-cffi

I think I will use this library to make a Python script that will rotate all images of a camera in a rosbag.
For the camera calibration, just decompress and recompress the stream with OpenCV.

closed 2016-09-25 by sbrodeur