sbrodeur/ros-icreate-bbb #8: Validate X-Y-Z axes on the robot

sbrodeur, 2016-09-14:
Define the  X-Y-Z axes on the robot and make sure this is coherent with odometry and IMU output.

sgcarrier, 2016-09-19:
Adjusted and standardized X-Y-Z axis. In self-reference, Y points straight ahead, X points right and Z points up. In orientation, Y points North, X points East, and Z points up. IMU madgwick (topic /imu/data) gives fairly accurate IMU info now.

sbrodeur, 2016-09-25:
I would like to see if we obtain something similar to Figure 7 in the link below after calibrating the magnetometer:
http://www.sensorsmag.com/sensors/motion-velocity-displacement/compensating-tilt-hard-iron-and-soft-iron-effects-6475

sgcarrier, 2016-09-26:
[image]

We suffer from both hard and soft distortion. Hard distortion is more present in the Y axis ( X axis has almost none). We also have the soft-distortion ellipse form. Note that this data was collected with the offsets I made.

I'm not sure if the Madgwick filter we use to fuse mag and acc data takes this soft distortion into account.

sbrodeur, 2016-09-27:
You may be right with this.
I think that if we plot the same data in 3D (X-Y-Z), we should be able to estimate the shift to the reference axes, and remove the bank and elevation angle effects. I can do that if you give me the data that was used to generate the figure, including the Z axis data.

sgcarrier, 2016-09-27:
[mag2.zip](https://github.com/sbrodeur/ros-icreate-bbb/files/495944/mag2.zip)
Here is the csv and spreadsheet file. The data different from the previous graph since my collected data was overwritten but it gives the same curves and all.

sbrodeur, 2016-09-27:
See script for calibration:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/imu/scripts/calibrate_mag.py

Usage:
```
python calibration_mag --input=magPoints_2.csv --display-data
```

It should display the data in 3D (navigable) and return the transformation (rotation matrix) required for calibration:
```
INFO:__main__:Using input cvs file: /home/simon/git/ros-icreate-bbb/data/magPoints_2.csv
INFO:__main__:X scale: -0.000020, 0.000018 (range of 0.000038)
INFO:__main__:Y scale: -0.000017, 0.000019 (range of 0.000036)
INFO:__main__:Z scale: -0.000041, -0.000040 (range of 0.000001)
INFO:__main__:Use this rotation matrix: 
 [[ 1.          0.          0.00988014]
 [ 0.          1.         -0.07183293]
 [-0.00978334  0.07740914  1.        ]]
```
[image]

sbrodeur, 2016-09-27:
The rotation matrix R should multiply the data x (i.e. xc =  R x) directly at the IMU capture. It can be hard-coded or passed as a ROS parameter to the driver node.

sgcarrier, 2016-09-27:
Applied corrections and this is the new graph
[image]

I applied the offset Y needed as well so its pretty centered. There is a slight ellipse shape. Do you think this is worth correcting?

sbrodeur, 2016-09-27:
See last commit efd77a5 for handling soft-iron effects.
This requires an additional rotation matrix Rxy and multiplicative factors fx and fy.
The idea to correct for the ellipse shape is to apply the rotation Rxy, rescale with fx and fy, then apply the inverse rotation Rxy^T (transposed of the rotation matrix) to get back to the original space.

```
INFO:__main__:Using input cvs file: /home/simon/git/ros-icreate-bbb/src/imu/scripts/magPoints_2.csv
INFO:__main__:X scale: -0.000019, 0.000018 (range of 0.000037)
INFO:__main__:Y scale: -0.000017, 0.000019 (range of 0.000037)
INFO:__main__:Z scale: -0.000041, -0.000040 (range of 0.000001)
INFO:__main__:Use this rotation matrix to correct for bank angle (z axis): 
 [[ 1.          0.          0.00988014]
 [ 0.          1.         -0.07183293]
 [-0.00978334  0.07740914  1.        ]]
INFO:__main__:Use this rotation matrix to correct for soft-iron effects (x and y axis): 
 [[-0.78891389  0.61450376  0.        ]
 [-0.61450376 -0.78891389  0.        ]
 [ 0.          0.          1.        ]]
INFO:__main__:Use these factors to correct for soft-iron effects (x-y plane): 
 [ 0.93339828  1.07683662]
INFO:__main__:All done.
```

[image]

sgcarrier, 2016-09-27:
[image]

The new correction works really well! The only thing is the offset on Y that I cant get right, but i think the reason for this is that I apply the offset before the rotations, when I should do it after. I will apply corrections and check it out.

sgcarrier, 2016-09-28:
[image]
This is the latest calibration. It's practically aligned perfectly and has a good circle shape.

sbrodeur, 2016-09-28:
Excellent! We can close the issue when the changes to the offsets will be committed.

sgcarrier, 2016-09-28:
Committed. For future reference : a8af559af217208a5e5bfa4e4b416b9da26a271b

sbrodeur, 2016-10-13:
While the IMU-calculated odometry follows the reference axes determined below, the odometry calculated from the internal robot sensors switches Y with X axes:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/irobot_create/bin/driver.py#L175

I am also able to see that in the visualizations.

sbrodeur, 2016-10-13:
See commit ee3e09d for a fix. This will need to be tested with the robot.

closed 2016-10-20 by sbrodeur