sbrodeur/ros-icreate-bbb #10: Test and optimize autonomous control mode

sbrodeur, 2016-09-19:
Make sure that autonomous control mode is still working fine:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/action/scripts/record/autonomous_control.py

The launch script that records the rosbag should also be adapted to the new topics:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/action/scripts/record/record_autonomous.launch

sgcarrier, 2016-09-27:
Autonomous control mode works. I had to switch angles from deg to Rad because of the new messages. Also changed the launch file for the new topics.

sbrodeur, 2016-09-28:
It seems that only the backward moving when hitting an object needs to be adjusted. The robot should move backward a fixed 3-5 cm (the minimum to clear the bumper).

sgcarrier, 2016-09-28:
Again it was an issue of units due to new messages. Changes were done here: 28443bce8ab5f3b4439f175c2a49934f886f263c

closed 2016-09-28 by sbrodeur