sbrodeur/ros-icreate-bbb #11: Test rosbag recording on robot

sgcarrier, 2016-09-23:
Verify that recording goes well, doesn't saturate and starts with appropriate key presses. Rosbags should be well labeled and allow multiple recordings without having to connect via ssh to change things.

sgcarrier, 2016-09-26:
Recording goes well, battery sustains fine. To start recording Y+B buttons are pressed and  records for up to 10 mins or X+A is pressed, whatever happens first.  

When saved they follow the format session_X were X starts at 0 and increments each time. However, everytime  the service restart, this is reset, thus care must be taken to transfer rosbags everytime a reboot is done before recording again or else the rosbags are overwritten

sbrodeur, 2016-09-27:
To solve the filename overwritting, I would just save the rosbag with the following format:
session\_\<datetime\>

See the command below for an example of \<datetime\>:
```
date +"%Y%m%d_%H%M%S"
```
This returns something like '20160926_210957' which can be appended to the filename to indicate the date and time of the recording.

sgcarrier, 2016-09-27:
Fixed. Output now in the format : session_[year][month][day]_[hour][min][sec].bag

closed 2016-09-28 by sbrodeur