sbrodeur/ros-icreate-bbb #12: Performance analysis

sgcarrier, 2016-09-27:
Investigate if all ressources are used correctly and that there are no flagrant bottlenecks that could impede our tasks. 

So far, notice that CPU usage skyrockets to 100% when I start rosbag recording. I have 2 possible explanations: 1 - Rosbag record takes a lot of processing power in all cases and my only recourse is to reduce the number of topics. 2 -  I'm making an error in my subprocess creation and deletion for recording and its causing performance issues (i could use valgrind to investigate this).

Aside from recording, i didn't see anything odd in the performance stats.

sbrodeur, 2016-09-27:
Note that recording a rosbag is particularly CPU-consuming when compression is activated. We should make sure that no compression is enabled.

sbrodeur, 2016-09-27:
The rosbag recording can be tuned a little with a few parameters on this line:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/action/scripts/record/remote_control.py#L251

- Suppress console output with --quiet
- Reduce the amount of RAM (in MB) for buffering with --buffsize=128,   and possibly even to 64MB.

Try with these additional options, and if this doesn't help, then we can start thinking about filtering some unneeded topics.

sbrodeur, 2016-09-28:
Validated and fine with the above modifications.

closed 2016-09-28 by sbrodeur