sbrodeur/ros-icreate-bbb #13: Validate control range of motor speeds

sbrodeur, 2016-09-30:
Ideally, with joystick control we should see smooth transitions in the velocities of the motors (e.g. analog mode), rather than a binary mode (e.g. null or full speed). It should look like this, and could even be smoother:

[image]

We just need to validate this by viewing the motor control signals in rqt, then calibrate the joystick if necessary.

sgcarrier, 2016-09-30:
[image]

It's a bit hard to tell now that joystick commands are sent asynchronously, but here is the graph. We see values are no longer binary (max, min) and can take many values in between.

sbrodeur, 2016-09-30:
This looks nice. What about plotting the same data, but from the topic /irobot_create/motors?
With this topic, we will see exactly what the robot executed at all time.

sgcarrier, 2016-09-30:
[image]
 
Heres the motor, for the same sequence. We see the shape is roughly the same even though motors has a synchronous output.

sbrodeur, 2016-09-30:
This is perfect! There is no problem here.

closed 2016-09-30 by sbrodeur