sbrodeur/ros-icreate-bbb #14: Finalize specifications for experiment I

sbrodeur, 2016-09-30:
Experiment 1:
The robot is moving around a room, controlled by the experimenter using a joystick. Each recorded session is approximately 15 min. The beginning and end of each session is cut and adjusted to that the length is exactly 10 min. There are 4 session recordings per room, with various starting points and trajectories. There is little to no moving objects (including humans) in the room. The robot is directed by the experimenter not to hit any obstacles.

Room 1: TO BE DETERMINED
Room 2: TO BE DETERMINED
Room 3: TO BE DETERMINED
Room 4: TO BE DETERMINED
Room 5: TO BE DETERMINED
Room 6: TO BE DETERMINED
Room 7: TO BE DETERMINED
Room 8: TO BE DETERMINED

This experiment will provide 32 sessions of 10 min each.

sbrodeur, 2016-09-30:
For this experiment, we need to find 8 rooms with different gist (e.g. computer lab, hallway, classroom, lunch room). It would be helpful to have some overall pictures of the rooms (i.e. to share with the dataset). The rooms should be large enough so that we can record 4 sessions of 10 min without the same path.

sbrodeur, 2016-09-30:
Here is a list of possible rooms compiled by @sgcarrier 

| Room  | Faculty | Human Activity | Description |
| ------------- | ------------- | ------------- | ------------- |
| C1-4113/4115 |  Eng. | None/low | Classroom with 5 large rectangular chest-high cubicles and front desk.  |
| C1-4127 |  Eng. | None | Big conference room with large center round table. Carpeted ground.  |
| C1-4016 |   Eng. | None/low | Classroom/auditorium with multiple level with inclinations instead of steps. Carpeted ground.  | 
| C1-5103 |   Eng. | None/low | More standard classroom with student desks.  |
| C1-5012 |   Eng. | Low | Conception room. Team work shop.  |
| C1-2045 |   Eng. | High/Very High | In front of student association and cafeteria. Many sofas and a lot of human activity.  |
| C1-3035/3041 |   Eng. | Low | Classroom like C1-4113/4115 but cubicles are waist high instead.  |

sgcarrier, 2016-09-30:
Room C1-4016 reservation confirmed. Administration says C1-4113/4115 and C1-4127 are not under their jurisdiction. Will need to find the place to contact.