sbrodeur/ros-icreate-bbb #15: Battery life

sgcarrier, 2016-09-30:
Currently at the 75 to 80% mark, the battery dies. This usually manifest as a sudden jump in the read value of charge to an absurd number ( 65535 or something). Its  -1, but since the expected value is unsigned it gives this maximal value of a 16-bit representation. Since we expect the robot to take into account his current charge in his learning, we should probably investigate this battery or find an alternative. 

Perhaps a cleaning cycle could help the battery or maybe replacing the battery by another or by normal rechargeable alkaline battery is the only way to go.

sbrodeur, 2016-09-30:
There is a nice forum here about hacking the APS battery:
<http://www.robotreviews.com/chat/viewtopic.php?p=77882>

sbrodeur, 2016-10-09:
New battery bought for the experiments. It should allow for longer lifetime.

closed 2016-10-09 by sbrodeur