sbrodeur/ros-icreate-bbb #17: Analyze missing messages (if any) in the recordings

sbrodeur, 2016-10-05:
Create a script that parses the rosbag files and estimate the number of missing messages based on the mean sampling rate, then output some statistics.

See the script below for an example of reading a rosbag file:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/action/scripts/record/data/convert_rosbag.py

sgcarrier, 2016-10-05:
Did some analysis with the new python script. Tested on S4,S3 and S2 of recordings in C1-4016. Note S4 was controlled and recorded from laptop. 

In all topics, less than 2% of packets  were considered 'dropped' (here dropped is that 2 messages had a distance of more than the mean plus 3 sigma).

Here is the histogram of the right video (which was being streamed at the same time as being recorded) of the S4 in room C1-4016. Black line is the mean, red are the standard deviation:
[image]

sbrodeur, 2016-10-06:
The bins are quite large, which makes the visual analysis difficult. Can you try with 50 times more bins?

sgcarrier, 2016-10-06:
Previous was 10 bins, heres with 300 bins
[image]

sbrodeur, 2016-10-06:
It would be nice to create a text report for each recorded session,  to store in a file aside the rosbag.
I was able to generate nice table with the Python package tabulate:
<https://pypi.python.org/pypi/tabulate>

```
from tabulate import tabulate
print tabulate([["/audio/left/raw", "10232", "10.23","2.31", "43"], ["/imu/data", "754", "5.73", "1.04", "13"]], headers=["Data","Total number of samples", "Average frequency rate in Hz", 'std', "Number of dropped samples"], tablefmt="grid", numalign="right", stralign="left", floatfmt=".2f")
```
This gives me this output:
```
+-----------------+---------------------------+--------------------------------+-------+-----------------------------+
| Data            |   Total number of samples |   Average frequency rate in Hz |   std |   Number of dropped samples |
+=================+===========================+================================+=======+=============================+
| /audio/left/raw |                     10232 |                          10.23 |  2.31 |                          43 |
+-----------------+---------------------------+--------------------------------+-------+-----------------------------+
| /imu/data       |                       754 |                           5.73 |  1.04 |                          13 |
+-----------------+---------------------------+--------------------------------+-------+-----------------------------+
```

sgcarrier, 2016-10-07:
Heres is an example file for S4

[E1_C14016_S4_20161004_180848.stat.txt](https://github.com/sbrodeur/ros-icreate-bbb/files/516337/E1_C14016_S4_20161004_180848.stat.txt)

sbrodeur, 2016-10-07:
I think we could remove the column 'mean' since it is redundant with the average rate, and the std deviation should be in Hz (i.e. same unit as average rate). We can also remove non-data topics such as: _/rosout, /rosout_agg, /tf, /irobot_create/cmd_raw, /video/left/camera_info, /video/right/camera_info._
There could be a flag passed to the analysis script to ignore those.
Those topics will be filtered out when exporting data into HDF5 format.

Is the average rate calculated from the timestamps of the rosbag (i.e. read_messages()), or the timestamps of the header (i.e. msg.header.stamp) for data messages?

How is the the number of  estimated dropped calculated?

sgcarrier, 2016-10-07:
Im using read_messages for the timestamps. Also, I estimate that a message have been dropped if the message took more than 1.1 times the mean than the mean time ( if time > mean+mean*1.1). I added the possibility to remove topics from command line arguments but im having trouble with std deviation in frequency. Will commit after thats solved.

sbrodeur, 2016-10-07:
We should use the capture timestamps (msg.header.stamp) if it is available, which is the case for all data topics.

sgcarrier, 2016-10-07:
Here is the graph for S4 using the messages' timestamps. It's almost too perfect and error never goes past 0.3% in all topics . A lot of topics have 0 errors too.

[image]

sbrodeur, 2016-10-08:
This is because the rosbag timestamps depend on the received time during recording, which can include buffering and network delay. The message timestamps from the headers indicate the capture time at the source, and are thus not affected by that. If it seems perfect than this means that the Beaglebone Black is not overloaded and is able to keep synchronous the data acquisition. This is good news.

sbrodeur, 2016-10-11:
Here is an output example with the latest script (see commit a32b723):

[image]

This is with a drop threshold of 1.0, meaning reject if period > 2 * average period.
Only the audio seems problematic, but since the recording depends on an internal buffer, timestamps may not be accurate. It should be ruled out whether this results in degraded audio.

closed 2016-10-12 by sbrodeur