sbrodeur/ros-icreate-bbb #21: Check rosout logs for errors

sbrodeur, 2016-10-11:
I'm seeing some possible errors in the rosbags, related to video and audio:

```
name: /video1
msg: [left] does not match name narrow_stereo/left in file /root/work/ros-icreate-bbb/src/camera/calibration/left_calibration.yaml
file: /root/build/ros_catkin_ws_kinetic/src/image_common/camera_info_manager/src/camera_info_manager.cpp
function: CameraInfoManager::loadCalibrationFile
line: 280
topics: ['/rosout', '/video/left/compressed', '/video/left/camera_info']
```
Same for the right camera

```
name: /mic2
msg: Unable to read audio stream!
file: capture.py
function: spinOnce
line: 106
topics: ['/audio/right/raw', '/rosout']
```
Same for the left microphone

It should be investigated if these are relevant or not to ignore.

sbrodeur, 2016-10-11:
It is possible to use rqt_console to have a quick view of the log messages:
[image]

Here is a log format (CSV) supported by rqt_console:
[log_rqt.zip](https://github.com/sbrodeur/ros-icreate-bbb/files/521828/log_rqt.zip)

I will write a short script to produce that CSV log file directly from a rosbag file.

sbrodeur, 2016-10-11:
See commit 866440d for the script to produce a CSV log file readable with rqt_console directly from a rosbag file. Note that rqt_console still requires a running roscore instance, so you may have to create a new instance manually on the host before running rqt_console.

sgcarrier, 2016-10-11:
Concerning the mic one, I see it pop up sometimes. I think its only a USB timing thing and it adjusts itself. If it appears, its always on service startup. I ignore it because all topics are created correctly and the error stops. Will still investigate to make sure.

sbrodeur, 2016-10-12:
For the camera calibration-related messages, I think we just need to rename the camera names in the YAML calibration files. Although these are just a warnings, I'm not sure if there is any effect.

sgcarrier, 2016-10-12:
Camera name in calibration files ajusted in 9540f936d55c866161e2d62997be640eeb5904fa

closed 2016-10-14 by sbrodeur