sbrodeur/ros-icreate-bbb #22: Blurry image quality from cameras

sbrodeur, 2016-10-11:
The images from the cameras seems anomaly blurry:
[image]

We may need to check the configuration here about exposure and gain:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/scripts/ros.sh#L19
The cameras also have auto-focus capability, but this is not configured. To disable autofocus:
```
v4l2-ctl --device=/dev/video6 --set-ctrl focus_auto=0
v4l2-ctl --device=/dev/video7 --set-ctrl focus_auto=0
```

To set the focus to a fixed value:
```
v4l2-ctl --device=/dev/video6 --set-ctrl focus_absolute=30
v4l2-ctl --device=/dev/video7 --set-ctrl focus_absolute=30
```

We may need to play with these options to see if there is any effect on the image quality. The exposure was set to a low value to avoid having blurry images during movement.

sgcarrier, 2016-10-11:
Commands dont seem to be recognized due to the focus_auto and focus_absolute keywords. 
Also, the gain and exposure commands, while work in command line seem to have no effect when changed in the ros.sh file

sbrodeur, 2016-10-12:
The ros.sh file is called when the ros service is started, so it requires to run the following command for changes to take effect:
```
service ros restart
```

sgcarrier, 2016-10-13:
Was able to set focus, exposure and manual mode in startup script (ros.sh), but the gain parameter is overwritten somewhere in the code. I suspect capturev4l2.cpp. Will continue to investigate.

sbrodeur, 2016-10-15:
We should be able to set exposure from the capture node:
http://stackoverflow.com/questions/26084794/v4l2-help-in-camera-settings

sbrodeur, 2016-10-15:
Here is some code example to integrate here:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/camera/src/capturev4l2.cpp#L158

```

        // Manual exposure control
        // See: https://linuxtv.org/downloads/v4l-dvb-apis/extended-controls.html
        struct v4l2_control stream_control_exposure;
        stream_control_exposure.id = V4L2_CID_EXPOSURE_AUTO;
        stream_control_exposure.value = V4L2_EXPOSURE_MANUAL;
        if(xioctl(_fd, VIDIOC_S_CTRL, &stream_control_exposure) != 0)
        {
        	perror("Couldn't set camera exposure to manual");
        	return 1;
        }

        // Manual exposure absolute value
        struct v4l2_control stream_control_exposure_value;
        stream_control_exposure_value.id = V4L2_CID_EXPOSURE_ABSOLUTE;
        stream_control_exposure_value.value = _exposure;
        if(xioctl(_fd, VIDIOC_S_CTRL, &stream_control_exposure_value) != 0)
        {
        	perror("Couldn't set camera exposure absolute value");
        	return 1;
        }

        // Manual focus
		struct v4l2_control stream_control_focus;
        stream_control_focus.id = V4L2_CID_FOCUS_AUTO;
        stream_control_focus.value = false;
		if(xioctl(_fd, VIDIOC_S_CTRL, &stream_control_focus) != 0)
		{
			perror("Couldn't set autofocus to manual");
			return 1;
		}

        // Manual focus absolute value
		struct v4l2_control stream_control_focus_value;
		stream_control_focus_value.id = V4L2_CID_FOCUS_ABSOLUTE;
		stream_control_focus_value.value = _focus;
		if(xioctl(_fd, VIDIOC_S_CTRL, &stream_control_focus_value) != 0)
		{
			perror("Couldn't set camera focus absolute value");
			return 1;
		}
```

sbrodeur, 2016-10-16:
See commit 92ffa7d for the integration (as above) of exposure and focus control. It needs to be tested on the robot.

sbrodeur, 2016-10-16:
I'm not sure if we can flip the image by the camera, since it doesn't seem supported:
```
root@beaglebone:~# v4l2-ctl --device=/dev/video6 --set-ctrl horizontal_flip=1
unknown control 'horizontal_flip'
```

sgcarrier, 2016-10-17:
So far, gain adjustment isn't working. The culprit _seems_ to be in initMmap(), when we start the capture. If we try to change setting after that it gives errors , and if done before they seem to be reset. So far the best solution seems to be adding a script that runs 30 sec after the end of ros.sh  that will run the settings command line.

sbrodeur, 2016-10-18:
See relevant discussion here:
<http://www.linuxquestions.org/questions/linux-software-2/camera-settings-in-linux-v4l2-or-uvc-drivers-4175525260/>

_"I finally solved the issue. The thing is that I had to change the Exposure and the Gain settings after I started the photo capturing not before (all other settings are to be set before). In addition Gain can only be set when Exposure is set to manual."_

sbrodeur, 2016-10-21:
It is possible to create a cron job to periodically (e.g. every minute) set the camera configuration:
```
* * * * * /root/work/ros-icreate-bbb/scripts/ros-cron.sh >> /root/work/ros-icreate-bbb/scripts/ros-cron.log 2>&1
```
This will currently execute every minute automatically on the robot.

Compared to the previous camera image, now the output is much more clear and still not blurry during movement:
[image]
The cameras seem also to have similar (albeit not perfectly equal) color balance output under the same configuration (e.g. check the wall color).

sbrodeur, 2016-10-23:
See commit 10e4455 for the cron job implementation.

closed 2016-10-23 by sbrodeur