sbrodeur/ros-icreate-bbb #24: Validate sampling rate of IMU sensor

sbrodeur, 2016-10-11:
The sampling rate for the IMU unit is currently at 20 Hz, which may be a little low. It would be nice to see if it could be increased to 100 Hz without much performance impact on the Beaglebone Black. However, this should only be done if we estimate that we are currently losing too much information at 20 Hz, specially for angular velocity and linear acceleration:

[image]

[image]

sgcarrier, 2016-10-11:
Maximum rate for the imu capture seems to be around 22Hz. Will continue to investigate if this is a performance issue that we can fix.

sgcarrier, 2016-10-11:
With new higher bitrate for i2c, and by using FIFO BYPASS mode for the devices, 130+Hz imu data can be obtained. 

However, FIFO BYPASS causes issues with the magnetometer. Also, having a sampling frequency over 20Hz seems to have too much of an impact on performance, especially during recording.

sbrodeur, 2016-10-15:
I just found linux drivers for the gyroscope (l3gd20), and accelerometer + magnetometer (lm303):

l3gd20
http://www.st.com/content/st_com/en/products/embedded-software/mems-and-sensors-software/drivers-for-mems/stsw-mems018.html

lm303
http://www.st.com/content/st_com/en/products/embedded-software/mems-and-sensors-software/drivers-for-mems/stsw-mems025.html

Those may allow for better sampling rate than the current implementation.

sbrodeur, 2016-10-16:
See commit 7a921d2 for all the necessary Linux drivers to install:
https://github.com/sbrodeur/ros-icreate-bbb/tree/master/src/imu/drivers

sbrodeur, 2016-10-16:
It seems that the LSM303D and L3GD20 sensors report data as HAL events:
https://github.com/01org/android-iio-sensors-hal
https://www.inovex.de/blog/android-sensor-integration-part-3-hal/

sgcarrier, 2016-10-16:
Right now, the captureBMP node for the barometer, is slower than its accel/gyro counter-part. I suspect this is due to the file reading which can be quite time consuming. I will continue to investigate to speed up the node. Once it reaches an acceptable speed, we could switch LMS303D and L3GD20 to this as well.

sbrodeur, 2016-10-17:
I think it is also possible that the delay comes from the overloaded I2C bus. All devices (RTC, accelerometer, gyroscope, magnetometer) are currently sharing that single bus.

sgcarrier, 2016-10-17:
captureBMP was using too expensive read operators on the files. Now it's running well and can go 45Hz+ if we desire.

sbrodeur, 2016-10-18:
See:
http://elinux.org/Sparkfun:_BMP085_Barometric_Pressure_Sensor#Reading_the_Temperature_and_Pressure_Programmatically

sbrodeur, 2016-10-18:
From the datasheet of BMP180 (page 13):

[image]
We should get higher rates, up to 128Hz.

sbrodeur, 2016-10-20:
Barometer will be updated at 15Hz.

closed 2016-10-20 by sbrodeur