sbrodeur/ros-icreate-bbb #25: Temperature measurements seem inaccurate

sbrodeur, 2016-10-13:
The temperature read from the IMU seems too high to be accurate:

[image]

We should check the current implementation using the LM303 sensor:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/imu/src/LMS303.cpp#L120

sbrodeur, 2016-10-13:
See commit 5dbd259 for a fix. This will need to be tested on the robot.

sbrodeur, 2016-10-14:
We may also use the temperature signal from the BMP180 sensors connected on the same I2C bus.
See here to compile the kernel driver module:
<https://github.com/bsapundzhiev/bmp085>

sgcarrier, 2016-10-14:
See commit 82ae0c1013cf0bd1e29f5fcb17c7944110dbfba4 for the corrections. 
The temperature was given in 12-bit , right-justified, 2s complement, and was calibrated to 25C.

sgcarrier, 2016-10-14:
Above procedure for BMP180 returns reasonable values, it could be pretty easy to add pressure and new temperature sensor. This new one would replace the old one that was more an internal temp probe for accelerometer and compass correction.

sgcarrier, 2016-10-14:
In commit b6c5b2c6f407a291d974cb74248c7650de594f22 added barometer node. It publishes atmo pressure and temperature. Temperature values seems much more reliable than the LM303 ones.

sbrodeur, 2016-10-15:
To have better temperature accuracy (and maybe pressure too), we only need to clear out the chip by removing the electrical tape that protects the PCB.

closed 2016-10-15 by sbrodeur