sbrodeur/ros-icreate-bbb #29: /Imu/data values never change

sgcarrier, 2016-10-18:
The values published in /imu/data for the orientation never change or change very slowly. My initial guess is that since /imu/data_raw and /imu/mag are on different nodes now, the Magwick filter can't find matching timestamps for both. Or, that the scale of one of the instruments has been reduced ( and thus the very slow movement).

sbrodeur, 2016-10-19:
The problem comes from the l3gd20 gyroscope driver, where the sensitivity is not applied and thus the results are not in udps as expected:
https://github.com/sbrodeur/ros-icreate-bbb/blob/master/src/imu/drivers/l3gd20/l3gd20.c#L667

I think this was commented out to avoid overflows.

sbrodeur, 2016-10-19:
The orientation problem is fixed with commit 8a986ce.
We only need to figure out why soft-iron correction doesn't work anymore (and was thus deactivated).

sbrodeur, 2016-10-20:
Tested and fixed with commit 4db406b.

closed 2016-10-20 by sbrodeur