sbrodeur/ros-icreate-bbb #35: Validate all sensors on the robot in real-time and batch mode

sbrodeur, 2016-10-24:
Use these commands to visualize each sensor:
```
# Audio (left and right)
python src/audio/nodes/viewer.py _input:=/audio/left/raw
python src/audio/nodes/viewer.py _input:=/audio/right/raw

# Camera (left and right)
rosrun camera camera_viewer _input:=/video/left
rosrun camera camera_viewer _input:=/video/right

# IMU (magnetometer, orientation and accelerometer/gyroscope)
python src/imu/nodes/viewer_mag.py _input:=/imu/mag
python src/imu/nodes/viewer_orientation.py _input:=/imu/data
python src/imu/nodes/viewer_acc_gyro.py _input:=/imu/data_raw

# Odometry
python src/irobot_create/nodes/viewer_odometry.py _input:=/irobot_create/odom
```

sgcarrier, 2016-10-28:
Madgwick filter in post processing seems to give incorrect values. For example, while the z axis should always stay straight up, it tend to move a drift a lot. Will need to investigate.

sgcarrier, 2016-10-28:
Turns out I was probably at fault for the previous Madgwick issues. However, we have to address the gain that is set to 0.1 in live mode and 0 in offline mode.

sbrodeur, 2016-10-30:
The gain should be set to 0.1 in offline mode. At 0.0, if I look at the equations, there is no filtering. This is why it is noisier in offline mode compared to real-time mode.

sgcarrier, 2016-10-31:
Gain increased to 0.1 in commit d0f07f1484c710d7ac7f4bd2c296cd2a40336ef5 .

closed 2016-11-01 by sbrodeur