sbrodeur/ros-icreate-bbb #37: Battery charge is always zero in visualizations

sbrodeur, 2016-11-02:
Not sure if the problem is that the battery charge is wrongly reported by the robot, or there is a bug in the visualization script. The rosbags should be checked first.

[image]

sgcarrier, 2016-11-02:
Here is the output of a rosbag recorded yesterday. 
header: 
  seq: 79874
  stamp: 
    secs: 1478037977
    nsecs: 240295485
  frame_id: base_link
voltage: 15.0839996338
current: -0.884999990463
charge: 2.23300004005
capacity: 2.70199990273
design_capacity: 3.0
percentage: 0.826424896717
power_supply_status: 3
power_supply_health: 0
power_supply_technology: 0
present: True
cell_voltage: []
location: ''
serial_number: ''
---

Ive seen the behaviour of 0 charge before. It usually happens when the battery is simply low and it floors the readings to 0. Although usually, the robot doesnt operate for long after that.

sbrodeur, 2016-11-02:
The problem was wrong unit conversion in visualization script. See commit a4364ab for bug fix.

closed 2016-11-02 by sbrodeur